orsjb/HappyBrackets #2: Beads dependency

samgwise, 2016-05-06:
To interact with Beads via dependency management it would be nice to have it in maven central.
See here on how to add it: [Guide to uploading artifacts to the Central Repository](https://maven.apache.org/guides/mini/guide-central-repository-upload.html)

Alternatively with Gradle we can continue to keep a jar in our project?
Another option is the sub-git approach, we could pull beads in as a sub-git and add it to our Gradle build.

@orsjb what are your thoughts?

orsjb, 2016-05-06:
I want to get Beads into Maven central. I’m not totally up to speed on the process of setting it up to build and deploy there. Perhaps we can go through this in person next time. One concern then is what to do with Beads dependencies that are not in Maven. On that matter, I note that both Pi4J and NetUtil are on Maven and seem to be quite up to date. These complete our main dependencies for HB.

samgwise, 2016-05-06:
OK, That sounds like a good plan. I'll put off adding beads properly until next week then.

orsjb, 2016-05-06:
Unless it is trivial for you to do, I can give you write access there. Beads is using an Ant build file with a bit of customisation to make the Processing and Java distributions. So I’m not sure if there should then be a separate Gradle/Maven build file. This is what has been holding me up. Probably best we talk it through.

samgwise, 2016-05-18:
As we've opted to bundle the pre-built library and it's dependencies in the project, I'm going to mark this as closed. We can revisit this if other options become available.

closed 2016-05-18 by samgwise