orsjb/HappyBrackets #5: Redesign device-controller communication and config sharing

orsjb, 2016-06-08:
Device/controller should never automatically send/receive config file data. This should only ever happen as the result of a user action.
Device always runs the config file it finds on startup.
This means we need another fix for the "no-zeroconf" scenario. New fix should be the following:
Device broadcasts to controller both its hostname and IP. 
Controller stores both, but only uses IP to communicate with device.

samgwise, 2016-06-08:
The automated config update behaviour appears to have been removed during the great refactor of 2016. This explains why it appeared to have stopped working last week :)

On to setting up the new address based communication.

samgwise, 2016-06-14:
Progressed on this issue today. Changed controller and device communications to use IP addresses instead of host names.

orsjb, 2016-12-03:
Resolved, I think.

closed 2016-12-03 by orsjb