orsjb/HappyBrackets #7: Raspberry Pi is not connecting the the given WiFi credentials

orsjb, 2016-07-06:
Tested on a D-Link. The Pi is available when connected via ethernet but simply fails to connect to this particular router. It might be because of something specific to the configuration of the router, so it would be good to know if this is working anywhere at the moment.

samgwise, 2016-07-06:
Is this via the OSC message  to /PI/config/wifi of the deivce? (See net.happybrackets.device.network.NetworkCommunication line 90-99 of commit 18b0e16)

orsjb, 2016-07-06:
No, I've not heard of this message before. 
The /etc/network/interfaces file is correct but the wifi is not connecting. I'm trying to manually set it with ifdown and ifup. Possibly it is a problem with my router.

orsjb, 2016-07-06:
Note: I don't know about these OSC messages. Where are we documenting them?
Also: it should be /device/ rather than /PI/.

samgwise, 2016-07-08:
I haven't fully tested the message behaviour yet so I haven't added it to any extant documentation. and the /PI/ was a typo, it has always been /device/.

orsjb, 2016-12-03:
Resolved.

closed 2016-12-03 by orsjb