orsjb/HappyBrackets #8: HappyBrackets IntelliJ plugin <-> device communication can't handle a change of WiFi network

orsjb, 2016-07-06:
If you are running the IntelliJ Plugin (Controller) and you are on WiFi network A, then you change to WiFi network B, then try to get a Pi to talk to the Controller then it does not connect. If you restart IntelliJ then it works.

Possible Reasons: Device is initialised when we first run the program. It finds the assigned IP address, hostname etc. I assume that we are then using IP addresses for communication, and these are changing when we change networks. However, it could also be something to do with the selected NetworkInterface, which is also assigned at runtime.

Solution (IP issue): We should avoid using IP addresses wherever possible and try to use hostnames for all communication. This should mean that it doesn't matter which network you're on and what your IP address is.

Solution (all issues?): We should constantly scan the network config and reset all comms components in response to changes. Currently our code is not organised well enough to do this. We can also have a "force refresh" option.

Workaround: restart IntelliJ if you change WiFi network.

orsjb, 2016-07-06:
Note to @samgwise : I believe we are using IP addresses in cases where Windows users do not have bonjour (zero-conf, via iTunes?) set up. If so, and if this is causing trouble, then we simply make it a requirement that this is set up.

samgwise, 2016-07-06:
I did some research tonight and found that we may have been doing some of the multicast networking incorrectly.
given this Oracle tutorial page:
http://docs.oracle.com/javase/tutorial/networking/datagrams/broadcasting.html

MulticastSockets are only for receiving multicast packets. For sending multicast sockets you send to the multicast address from a normal socket. In our case we should send from a DatagramSocket. See this Oracle reference: http://download.java.net/jdk7/archive/b123/docs/api/java/net/DatagramSocket.html

I've corrected Synchronizer, ControllerAdvertiser and ControllerDiscoverer to use DatagramSocket for sending. This appears to remove the need to set an interface in my basic tests on linux with no zero-conf MDNS setup.

orsjb, 2016-07-06:
@samgwise I think we need to catch up to discuss this. This was the basis for my revised BroadcastManager that I had working nicely the other day. However, I couldn't see a way to run the listener without requiring to specify a network interface. 

I changed your Synchronizer code because it was using ControllerConfig, whereas we need to use LoadableConfig. I also had to change the DatagramSocket sender because we were having an error of the socket (or port, or address?) already being in use.

orsjb, 2016-12-03:
Resolved. Device now aggressively checks status of network interfaces and does full rebuild of any network interfaces that change.

closed 2016-12-03 by orsjb