orsjb/HappyBrackets #11: IntelliJ plugin does not read controller-config file

orsjb, 2016-07-20:
Devious IntelliJ plugin. Currently we can find the controller-config when we're running in sandbox, but the REAL plugin does not find it. The controller-config and other resources (as stored in the project's resources directory) actually gets jarred in the final distro, which is what you'd expect. Problem is we're not reading those files in the correct way. Need to check best practice here.

orsjb, 2016-12-03:
@OliverColeman can you advise if this is resolved?

OliverColeman, 2016-12-06:
This is resolved in that on first load the configuration text area is populated with the configuration file rolled into the jar file. If the configuration is saved or loaded to/from another file then that file will be loaded the next time the plugin is loaded.

closed 2016-12-06 by OliverColeman