orsjb/HappyBrackets #12: .gitignore mess

orsjb, 2016-07-26:
Lots of files being ignored that we don't want to ignore. 
The HappyBrackets Project config folder entirely missing. The HappyBrackets Project IML file missing. Some stuff in Distribution missing.

samgwise, 2016-07-26:
We probably need to refactor how we lay out our .gitignore files.

We should have one .gitignore per project tree. I think some of these problems may be from copying .gitignore files from the HappyBrackets project directory into the distribution tree.

For example there is a .gitignore file in the HappyBrackets/config directory which ignores *.json, this behaviour allows us to have the default files which are then overridden by user created .json files and the user created files will not be committed and will not clobber anyone else's config files.

orsjb, 2016-12-03:
Done a clean up. Prob not perfect still, but clean at least.

closed 2016-12-03 by orsjb