orsjb/HappyBrackets #20: Raspberry Pi is not connecting over ethernet

orsjb, 2016-12-02:
Setup: WiFi router with different SSID (not PINet). Pi is connecting over ethernet. Computer is connecting over WiFi. The Pi is showing up as a DHCP client but is not being seen by the computer. It seems likely this is something running on the Pi that is thrashing the WiFi interfaces and blocking other processes or network comms.

samgwise, 2016-12-02:
If it has connected over a wired connection it seems odd that the PI's WiFi activity would effect other devices. It may be a time to live limitation on packets?
Alternatively there is the WiFi reconnect script on the PI which has not been tested on a wired PI in a while, this may also be causing errors.
If the PI may have difficulty picking a host name if the WiFi is not connected (This could be an old solved bug though, I can't quite remember).

If the PI has appeared in the DHCP logs can you ping it's IP from your laptop?
If that is successful try connecting via the IP with SSH `ssh <username>@<IP>` .
If you can log in try checking the host name, processes and logs; the usual debugging steps.

orsjb, 2016-12-04:
I just ran this test again and it worked. It is slow to connect, but this time it is working perfectly. I'll leave this issue open for a couple more days and try to re-test under different circumstances. Otherwise perhaps just a furphy.

angelofraietta, 2017-10-12:
The network code has been updated

closed 2017-10-12 by angelofraietta