orsjb/HappyBrackets #21: Broadcast listeners are also receiving /device/alive messages

orsjb, 2016-12-02:
Broadcast listeners are also receiving /device/alive messages. Does this mean they're listening all on the same channel? Is /device/alive now being broadcast? That would make sense and if so this is not necessarily an issue. Only if that's a mistake. Still it might be sensible to have a different channel for these messages.

samgwise, 2016-12-02:
Yes I believe that is the correct behavior for the current design.

orsjb, 2016-12-03:
I've added an 'else' clause to so that only messages not already consumed by the DeviceConnection manager get forwarded to listeners.

orsjb, 2016-12-03:
Sorry that should read the net.happybrackets.device.network.NetworkCommunication class.

closed 2016-12-03 by orsjb