orsjb/HappyBrackets #23: Add password protection to controller->device communication

orsjb, 2016-12-03:
Need some basic level of protection so that a device is not vulnerable to attacks from a fake server. 

Simple version: unencrypted passphrase stored in both the device's and controller's config files and send by unicast. Passphrase must match in order for any HBActions to be run.

This could also be used by the device to determine what server to talk to. Makes it much easier to have multiple device/server combinations on the same network.

orsjb, 2016-12-04:
Strategy:
* Presently the class is sent over as a byte array. 
* From the SendToDevice class: Append this byte array with (1) an int indicating the length of the following string, (2) a string, of length specified by the previous int. 
* In the HB class: Parse the byte array to obtain the string, leaving the original class byte array.

The string should be a passphrase specified in both the device and controller config files. 

This is still not a good strategy. It would be easy to spoof being a device in order to learn the passphrase from the controller. Though an additional security measure would be to only allow communication with certain devices by the controller.

Alternative?

samgwise, 2016-12-05:
If you are opting for a pre-shared key approach it would be more effective as you have said if the password was not broadcast.
We could improve the proposed strategy by encrypting the payload with a cipher, perhaps Blowfish2 or Rijndael. We could prefix a fixed width unidirectional hashing function to provide a checksum (SHA-2 might be a good candidate). 
Additionally we should transmit these payloads over TCP sockets to provide better assurance that the entire payload arrives at the host and can be ordered correctly (especially important for WiFi!).

In this strategy the server calculates the checksum of the class file and begins the Byte array with it (I think this might end up being 64 Bytes for a fixed width SHA-256?).
Then the server can encrypt the class file to send across, adding the encrypted version to the Byte array.
When the device receives the complete payload it attempts to decode the content contained after the fixed width hash. When the content has been decoded according to the devices key it constructs it's own hash of it's decoded content and compares the hash it has generated to the hash contained in the initial chunk of the byte array (the hash from the server).
If the two hashs are the same then the content has been decoded successfully and as such must have been passed from a controller with the same key as the device.

This strategy clearly depend upon the pre-shared key being private however the key is not shared during transactions and as such is not particularly susceptible to man in the middle attacks and general network snooping.

orsjb, 2016-12-06:
Nice. This sounds great.

Here's another simple and in many ways more robust measure, which would combine well with the encryption. 

HB.jar runs with three modes: 
* open: can receive incoming code, still using handshake or encryption or whatnot.
* local: only allows incoming code from localhost.
* closed: not listening at all for incoming code.

The default would be "local", meaning that when you run HB.jar from your local machine you are not opening it up to compromise. The run.sh script would default to setting access to "open". However, when you were doing a production install or whatever, you could set your action up on the device and set the access to "closed".

OliverColeman, 2017-01-12:
I've just pushed changes that introduce access modes and encryption of class files (along with using a hash to verify decrypted class files). The encryption is "AES/CBC/PKCS5Padding" (AES is a standardised subset of Rijndael). The encryption stuff is tucked away in a new class, net.happybrackets.core.Encryption. There's a new parameter for the controller and device config files, 'encryptionKey', to allow specifying the key used to encrypt the class files (defaults to "abc123").

closed 2017-01-15 by OliverColeman