orsjb/HappyBrackets #25: Misc issues and feature requests raised at workshop

orsjb, 2016-12-05:
* Sending code didn’t work, just at one point during the entire day. I think this was possibly due to changing *project* whilst in IntelliJ (i.e., open Project A, load plugin, then open Project B, then load plugin again - plugin seems to get the right path to the compiled classes, but didn’t send anything over).
* Sometimes massive hang or freeze when loading plugin. Also occasional spinning wheels of doom at various points. Not a big problem, but quite clunky.
* Plugin GUI elements and text appear TINY! in Windows. Maybe if we just knew where the user can zoom on these then that would be good (or maybe we need to implement our own zoom behaviour over the entire GUI?).
* Sometimes plugin devices list resets. I think that when an element is added to the list this might prompt the list to be rebuilt and the status messages get reset to “status unknown”.
* Devices list should permanently show the device ID. (And allow you to manually set it).
* Requested: option to send code to one specific device.
* Device should ‘latch’ onto a controller and not allow itself to connect to another controller.
* Requested: kind of auto-send behaviour so that when the file is changed it automatically sends. And/or a key command that sends the currently open file (how does the plugin know which is the currently open file? Presumably an action event is created with the file as an argument).
* Add behaviour to the plugin to just ‘flash’ the current compiled code to the device.
* Plugin should auto-reload the default (or last) known_devices file when it opens up.
* The list item height should probably be fixed.
* Request: be able to make more than 1 simulated Pi (that will require quite a bit of reimplementation).
* Also a note: we are currently hashing devices by their device name (= usually hostname). This will be problematic whenever devices do not have unique device names.

hokey, 2017-04-13:
Is there any known work around to the plugin beach balling/freezing IntelliJ?  It's happening on two different macs for me.

orsjb, 2017-04-16:
Hi Mike,

can you give any more info on OS, version of Java, version of IntelliJ? Any other plugins you’re running?
We’ll see what we can do.

Ollie

hokey, 2017-04-17:
Hi Ollie,

Here is the info from my work Mac:

[image]

IntelliJ IDEA 2016.3.4
Build #IC-163.12024.16, built on January 31, 2017
JRE: 1.8.0_112-release-408-b6 x86_64
JVM: OpenJDK 64-Bit Server VM by JetBrains s.r.o

Plugins:
[image]

This setup gives me a beach ball and I have to force quit IntelliJ.  I'll get my home Mac info together later tonight.  Thanks for checking this out,

-Mike

hokey, 2017-04-19:
Hello,

Here are my home computer's stats:
[image]
IntelliJ IDEA 2017.1.1
Build #IC-171.4073.35, built on April 6, 2017
JRE: 1.8.0_112-release-736-b16 x86_64
JVM: OpenJDK 64-Bit Server VM by JetBrains s.r.o
Mac OS X 10.12.4
[image]

orsjb, 2017-04-21:
Thanks Mike,

we’re looking into this, but I can confirm that we’ve not come across this anywhere else. So I’m wondering if it is either a plugin clash. Are you using a non-basic IntelliJ configuration?

Ollie

hokey, 2017-04-24:
I'm using the CC edition.  I have brew installed, that may muck things up.  I'm not sure what the cause is.

​I ended up moving everything over to an Ubuntu VM and got the plugin to launch last night.  I was getting no audio before but the move from 1.8.0_121 to 1.8.0_131 resolved that issue for me.  I tried updating on the mac as well, but was unsuccessful.  I'm going to move forward with the VM for the next coding project.   Thanks for assisting me.

orsjb, 2018-02-15:
Massive overhaul by Angelo has probably resolved these.

closed 2018-02-15 by orsjb