orsjb/HappyBrackets #28: Plugin / GUI feature wishlist

orsjb, 2017-09-05:
The IntelliJ setup instructions are still too hard to follow: setting up JDK and linking to project.
Improvements to plugin interface elements, e.g., a sync button, a pop-up menu on individual devices.
Ability to copy the name of the device.
A button that would pop up an SSH terminal logging into each device.
GUI method to copy sound and classes over to devices.
A button that takes the current sketch and makes it auto-run.

angelofraietta, 2017-09-27:
Need to examine the singleton nature of plugin communication. A single version is running on from rats instance of tool window

angelofraietta, 2017-09-27:
A button that would pop up an SSH terminal logging into each device. - This has been implemented as a copy ssh command to clipboard