orsjb/HappyBrackets #29: Sensor library improvements

orsjb, 2017-09-05:
MiniMu update speed. 
In general for each sensor listener it should be possible to set the polling rate.
Improve library structure so that it is easier to grab a sensor and grab specific sensor values. e.g., give me a listener, make it listen to the sensehat, give me accelerometer data.
MiniMu code needs to be brought up to date. Reporting incorrect values.
Sensor names are obscure: can we just have a SenseHat listener? Or can we provide a lookup table for the sensehat elements?

angelofraietta, 2017-12-05:
I suggest we modify this so we do a detect on what hardware we have and use a general accelerometer, gyroscope, and magnetometer listener. We can detect what we have connected and call accordingly.

This will also allow us to simulate these things by creating virtual devices on debugger or workstation instance

angelofraietta, 2017-12-11:
Accelerometer, Gyroscope and Magentometer are done

angelofraietta, 2018-01-03:
We need to add some sort of check if we attempt to load specific LSM9DS1 and Accelerometer (or Gyroscope or Magnetometer) at the same time. Currently we an use generic or specific, but not both. Eg you can do either:
`Accelerometer mySensor = (Accelerometer) hb.getSensor(Accelerometer.class);`
or
`LSM9DS1 mySensor = (LSM9DS1) hb.getSensor(LSM9DS1.class);`

but not both.  We just need to add a test inside hb.getSensor to check if Accelerometer has loaded first, that LSM9DS1 will be loaded if user calls  (LSM9DS1) hb.getSensor(LSM9DS1.class);