orsjb/HappyBrackets #31: Fixes to project structure

orsjb, 2017-09-05:
The HB Project structure could be made more simple and generic so that it is easier to duplicate it and start your own hacks.
Generalise the teaching materials and decouple them from the Kadenze course: at present there is little help in understanding anything outside of doing the Kadenze course.

orsjb, 2017-09-05:
One possible solution is to have basically a blank project structure, or a stripped down one with a small handful of examples, and then to have a separate download of source code for the Kadenze course that can be imported. Possibly better just to arrange the packages so that the examples are sorted by topic, not by week, and the course is kept separate.

angelofraietta, 2017-09-28:
Need to add package org.apache.commons.lang to gradle so we can use StringUtils

angelofraietta, 2017-09-28:
Need to add script in Gradle to add date to module BuildVersion