orsjb/HappyBrackets #57: General exception handling

orsjb, 2018-01-14:
Need to ensure that the user always sees Exceptions. We should ensure as best we can that all the threads set up by HB are catching and reporting exceptions. e.g., Currently an exception that occurs in a SensorListener's sensorUpdated() method does not get reported.

angelofraietta, 2018-01-14:
Can you please give an example of the code that caused it?

I suspect I can put it in Sensor

 ```
   protected void notifyListeners() {
        Iterator var1 = this.listeners.iterator();

        while(var1.hasNext()) {
            SensorUpdateListener listener = (SensorUpdateListener)var1.next();
            listener.sensorUpdated();
        }

    }
```

angelofraietta, 2018-01-14:
in HB we should also deprecate 

`public Sensor getSensor(Class sensorClass) {`

and add one that throws null pinter exception to say we were not able to create

That way we do not need to check for sensor
```

 Accelerometer sensor = (Accelerometer)hb.getSensor(Accelerometer.class);
        if (sensor != null){ // not needed any more and they will have to add catch by compiler
```