orsjb/HappyBrackets #63: printStackTrace() is not outputting to the debug console

orsjb, 2018-02-15:
Run the following code in an HBAction:
 
```
        try {
            String x = null;
            String y = x.trim();
        } catch (Exception e) {
            System.out.println("problem " + e.getMessage());
            e.printStackTrace();
        }
```
Look in the device console. The only thing reported is "problem null".

angelofraietta, 2018-02-15:
Change the line in the run.sh so it has following (note addition of 2>&1)

(/usr/bin/sudo /usr/bin/java -cp data/classes:HB.jar -Xmx512m net.happybrackets.device.DeviceMain buf=$BUF sr=$SR bits=$BITS ins=$INS outs=$OUTS start=$AUTOSTART access=$ACCESSMODE $ACTION > stdout 2>&1) &

I have pushed it to git

Thanks

angelofraietta, 2018-02-16:
This is fixed. I have uploaded a new image also
2.0.1

closed 2018-02-16 by angelofraietta