orsjb/HappyBrackets #75: Add device aware global variables

angelofraietta, 2019-08-04:
Create global aware variables that can be added to HB instance. This will allow us to store variables across different sketches and same ones.

Use a registration mechanism with a name

HB.registerVariable(String variable_name, Object variable_val);
Object getVariable(String variable_name);
HB.updateVariable(String variable_name, Object variable_val);
HB.deleteVariable(String variable_name);

Use mapped strings to access

angelofraietta, 2019-08-04:
Looks like it already exists in share.

angelofraietta, 2019-08-04:
It appears these are cleared on reset. We will create register

angelofraietta, 2019-08-05:
Added in 4.2.0
Also added DataSmoother at the same time

closed 2019-08-05 by angelofraietta

orsjb, 2019-08-07:
I think that’s was mainly about the convenience wrapper functions in HB.
Yes, perhaps these should not be cleared on reset. Instead have a seperate reset for resetting variables.