orsjb/HappyBrackets #79: Renderer API features

orsjb, 2021-04-06:
Refactor package structure, remove SynchronisedModel. Replace with something sensible.
Remove need to access RendererController from within the sketch code. For example, instead of calling rc.displayColour(), you should be able to call displayColour().
You should not need to call rc.pushColour(). This should be done by the RendererController. The RendererController should make sure that all of the Renderer tick() calls occur and then it should complete the clock cycle with the push command.

orsjb, 2021-04-06:
Add documentation with examples (especially how to use the annotations).

orsjb, 2021-04-06:
In addition I think Renderer should give access to the "rc" instance, so that we can query things. I also think it should give access to the "hb" instance. This gives us access to the random number generator and other features.

gutosantos82, 2021-05-19:
These issues have been solved in the latest version.

closed 2021-05-19 by gutosantos82