orsjb/HappyBrackets #85: DeviceStratumSetTest fails

jacobsimionato, 2023-07-04:
When I run `gradle deploy`, I get a single test failure in `Class net.happybrackets.core.scheduling.DeviceStratumSetTest`:
```
java.lang.AssertionError
	at net.happybrackets.core.scheduling.DeviceStratumSetTest.testStratumAdjust(DeviceStratumSetTest.java:27)
	at java.base/jdk.internal.reflect.DirectMethodHandleAccessor.invoke(DirectMethodHandleAccessor.java:104)
	at java.base/java.lang.reflect.Method.invoke(Method.java:578)
	at org.junit.runners.model.FrameworkMethod$1.runReflectiveCall(FrameworkMethod.java:59)
	at org.junit.internal.runners.model.ReflectiveCallable.run(ReflectiveCallable.java:12)
	at org.junit.runners.model.FrameworkMethod.invokeExplosively(FrameworkMethod.java:56)
	at org.junit.internal.runners.statements.InvokeMethod.evaluate(InvokeMethod.java:17)
	at org.junit.runners.ParentRunner$3.evaluate(ParentRunner.java:306)
	at org.junit.runners.BlockJUnit4ClassRunner$1.evaluate(BlockJUnit4ClassRunner.java:100)
	at org.junit.runners.ParentRunner.runLeaf(ParentRunner.java:366)
	at org.junit.runners.BlockJUnit4ClassRunner.runChild(BlockJUnit4ClassRunner.java:103)
	at org.junit.runners.BlockJUnit4ClassRunner.runChild(BlockJUnit4ClassRunner.java:63)
```

I can work around this by commenting out that assertion for now, but let's investigate this and figure out if it's a real failure and related to upgrading some of the dependency versions, or broken for some other reason.

angelofraietta, 2023-07-26:
Just one thing on this. Does it fail every time or just sometimes?

-- 
-- 
Dr Angelo Fraietta
A.Eng, A.Mus.A, BA(Hons), PhD.

Home Page
https://www.newcastleguitarlessons.com.au/
https://orcid.org/0000-0002-3171-8491

There are those who seek knowledge for the sake of knowledge - that is
CURIOSITY
There are those who seek knowledge to be known by others - that is VANITY
There are those who seek knowledge in order to serve - that is LOVE
    Bernard of Clairvaux (1090 - 1153)

jacobsimionato, 2023-07-26:
I eventually realised if I run it a few times, it passes - just as your guide says :-D. Apologies for the noise!

closed 2023-07-26 by jacobsimionato

angelofraietta, 2023-07-26:
The issue is the way the jvm works.  I will give you a detailed example of
what is happening.

-- 
-- 
Dr Angelo Fraietta
A.Eng, A.Mus.A, BA(Hons), PhD.

Home Page
https://www.newcastleguitarlessons.com.au/
https://orcid.org/0000-0002-3171-8491

There are those who seek knowledge for the sake of knowledge - that is
CURIOSITY
There are those who seek knowledge to be known by others - that is VANITY
There are those who seek knowledge in order to serve - that is LOVE
    Bernard of Clairvaux (1090 - 1153)

angelofraietta, 2023-08-30:
This is what I believe the issue is.

What happens in the JVM is that they use what is called the Just In Time
(JIT) compiler.  What this does is parse the compiled java byte code and
turn that into machine code to execute. When it has done this, it caches
the machine code and it does not need to compile it again. So what i did
for code that i needed ti run in time was execute it a few times to cache
the machine code.  I mentioned it one of my papers.  You were really able
to experience when you first downloaded a patch to happy brackets that
started playing sound. It jitters at first. If you pause and then restart
the jitter goes.

I saw a very similar thing withe the synchniser code would fail tests. I
was also able to verify the issue. So did the same task multiple times to
get timer code compiled and then did timing tests.

-- 
-- 
Dr Angelo Fraietta
A.Eng, A.Mus.A, BA(Hons), PhD.

Home Page
https://www.newcastleguitarlessons.com.au/
https://orcid.org/0000-0002-3171-8491

There are those who seek knowledge for the sake of knowledge - that is
CURIOSITY
There are those who seek knowledge to be known by others - that is VANITY
There are those who seek knowledge in order to serve - that is LOVE
    Bernard of Clairvaux (1090 - 1153)