orsjb/HappyBrackets #87: Decouple HappyBrackets project from Beads project

jacobsimionato, 2023-07-19:
Currently, to build the HappyBrackets intellij plugin, you seem to need to clone the [Beads](https://github.com/orsjb/beads/tree/master) and Happy Brackets repos adjacent to each other, run the Beads `deployJarAndJavaDocToHappyBrackets` Gradle task in the Beads project, then try to build Happy Brackets.

This is a little finicky and I maybe unusual? It took me a while to figure out anyway. I think it'd be great if you could clone the HappyBrackets project and build it independently. I think we could achieve this by one of:
- Check the beads jars into the Happy Brackets git
- Upload Beads to some package repository like Maven and then have the Happy Brackets project depend on it that way.

angelofraietta, 2023-07-19:
Do them as dependent actions in gradle. Just order dependencies

-- 
-- 
Dr Angelo Fraietta
A.Eng, A.Mus.A, BA(Hons), PhD.

Home Page
https://www.newcastleguitarlessons.com.au/
https://orcid.org/0000-0002-3171-8491

There are those who seek knowledge for the sake of knowledge - that is
CURIOSITY
There are those who seek knowledge to be known by others - that is VANITY
There are those who seek knowledge in order to serve - that is LOVE
    Bernard of Clairvaux (1090 - 1153)

jacobsimionato, 2023-07-19:
Okay cool! Thanks so much for your help, Angelo!

Ahhh I see the Beads dependency is already declared here: https://github.com/orsjb/HappyBrackets/blob/6b152eb3ab83da3039085a2fae775cda1b8584db/HappyBrackets/build.gradle#L50C99-L50C99

And it's at Maven here: https://central.sonatype.com/artifact/net.beadsproject/beads/3.2

So I guess I'll debug why it didn't seem to download properly for me and I had to build locally.

angelofraietta, 2023-07-20:
When i did it i built beads separately because there were features that we
added that Ollie had not uploaded to maben

-- 
-- 
Dr Angelo Fraietta
A.Eng, A.Mus.A, BA(Hons), PhD.

Home Page
https://www.newcastleguitarlessons.com.au/
https://orcid.org/0000-0002-3171-8491

There are those who seek knowledge for the sake of knowledge - that is
CURIOSITY
There are those who seek knowledge to be known by others - that is VANITY
There are those who seek knowledge in order to serve - that is LOVE
    Bernard of Clairvaux (1090 - 1153)

jacobsimionato, 2023-07-26:
Sounds great - thanks for the guidance! Now I see this is configured in `HappyBrackets/IntelliJ Plugin/HappyBrackets_IntelliJ_Plugin.iml`.